odtu/EE464 #1: DCM in Cuk Converter

ghandeb, 2018-03-15:
@ozank 

Inductance değerini çok düşürmek converterı DCM'e sokar mı hocam?
Veya inductance üzerinde sabit akım olmasını voltajın sıfırlanmasını nasıl açıklarız?

Bu da böyle ramazan sohbetleri sorusu gibi oldu ama.

MelisaSarici, 2018-03-16:
@ozank 
Inductance değeri çok düştüğünde DCM olmuyor da akım ters yöne akmaya başlıyor. Inductance değerini de bu akım hep pozitif kalacak şekilde seçince Vout hiç 15 V olmuyor?

ozank, 2018-03-16:
Düşük inductance değeri o sorunu yaratır. Uygun bir değer seçmeniz gerekiyor. Hızlıca bir tasarım yapmak isterseniz Plex'de kullanabilirsiniz
https://www.plexim.com/academy/power-electronics/cuk-conv

Faydalı olabilecek bir kaç link
http://www.ese.iitb.ac.in/~suryad/lectures/EN206/Lecture-DC-DCConverter-2.pdf

closed 2019-03-21 by gkhnckl