odtu/EE464 #2: Vout

MelisaSarici, 2018-03-16:
@ozank 
Hocam, teorik olarak hesap edilen Duty cycle verildiğinde Vout değerinin 15 V değil de 14 V gibi bir değer çıkması normal mi?

ghandeb, 2018-03-16:
@ozank 
bizde de oldu o

ozank, 2018-03-16:
baktığınız varsayıyorum ama switch ve diyotlar ideal de mi? Üstünde bir gerilim düşmesi olmuyor?

ozank, 2018-03-16:
Bi de çıkıştaki L-C filtrenin karakteristiği ve anahtarlama frekansının ilişkisine de dikkat edin.

berkzorlutuna, 2018-03-16:
Switch ideal değil Ozan hocam. Switch direncini 0 yapmaya çalıştık fakat program izin vermedi. 0.001 ohm olarak değer verdik switch direncine. 

iPhone’umdan gönderildi

Mertkaracelebi, 2018-03-17:
Hocam en ideal halini kurdum devrenin output voltajının 14.96 V olması sorun değildir umarım. Aynı zamanda kullandığımız load constant power load mu olması lazım yoksa constant impedance yük işimizi görür mü ?

ozank, 2018-03-19:
Sabit direnc uygundur Mert.

closed 2018-03-21 by ozank