odtu/EE464 #12: core loss?

MERTAYDIN46, 2020-03-29:
hocam merhaba,
steinmetz's equation da frekansın hem core lossu hem de eddy lossu etkilediğinden equationa eklendiğinden bahsetmiştik fakat ericson's transformer design notunda frekansı hiç katmamış işin içine? ben tam olarak neyi ihmal ettiğinide anlamadım. yardımcı olursanız çok sevinirim.
iyi günler.

[image]

tokgozfurkan, 2020-03-29:
Merhabalar Mert,
Frekansı hesaba katmış aslına bakarsan. Yolladığın formülün birkaç sayfa ilerisinde aşağıdaki gibi bir şey demiş. Yani frekans aslında formülün içinde var ama K ve beta katsayıları olarak varlar. Bana kalırsa biraz karışık anlatmış transformer tasarımını. Diğer kaynaklara bakmayı deneyebilirsin veya üreticinin design guidelarına da bakabilirsin. 

[image]

MERTAYDIN46, 2020-03-29:
teşekkürler 
bir sorum daha vardı da yine aynı equation da delta_B değeri sanki sadece positive region da salınım yaparken verilmiş fakat B değeri eksiye simetrik şekilde düşünce (push pull daki gibi) positive delta_B nin yarısını equationa koyup bulduğumuz sonucu 2 ile çarpacağız gibi hissettim yanılıyor muyum acaba?

tokgozfurkan, 2020-03-30:
Merhabalar Mert,
Örnekte izole Cuk converter yapıyor. Burda trafo sadece pozitif cycleda çalıştığı için B-H curveün pozitif tarafını tarıyor. Ama push-pull gibi simetrik bir converter olsaydı hem negatif hem pozitif tarafta çalışacaktı. Push-pull örneğinde pozitif cycledaki core lossu bulup ikiyle çarpmak toplam core lossa eşit oluyor.

closed 2020-04-02 by tokgozfurkan