odtu/EE464 #15: Transformer sarım sayısı

burakkemalkara, 2020-04-04:
2. ödevin 3. sorusunun a şıkkında trafonun sarım değerleri sorulmuş fakat kablo seçimi yapmadan sarım sayısını nasıl hesaplayabiliriz anlayamadım. Ancak minimum sarım sayısını nüvenin satüre olmayacağı nokta bazında bulabilrim. B şıkkında da kablo seçmemiz istenmiş oradan da maksimum sarım sayısını bulurum. Optimum sayıyı bulmam için kablo seçimini yapıp hesaplamam gerekmez mi?

tokgozfurkan, 2020-04-04:
Merhabalar Burak,
Sarım sayısı N.I değerini değiştirdiği için operating flux densityi ve dolasıyla core loss değerini değiştirir. Ek olarak magnetizing inductance değerini de belirler. O yüzden operating flux density seçip daha sonrasında sarım sayısını bulabilirsin. Trafo tasarımında farklı metodlar var. Anladığım kadarıyla sen windows areayı tamamen dolduracak şekilde tasarıma başlıyorsun. Kablo kalınlığı bu yüzden lazım oldu sana. İlla bu metodla yapmak istiyorsan önce kabloyu seçebilirsin. Ama dediğim gibi tek yol o değil. Manyetik tasarımda bir sürü parametre var. Her metod farklı bir parametreden başlıyor hesaba.

burakkemalkara, 2020-04-04:
Hocam amacım maximum ve minimum sarım sayılarını ve kablo direncini bilip Core loss ve copper loss toplamımın minimum olduğu sarım sayısını bulabilmekti. Önce operating flux density seçersem ona bağlı seçmiş olduğum sarım sayısı nüvenin içine sığmazsa işlemin başına dönmem gerekmez mi

tokgozfurkan, 2020-04-04:
Evet o yüzden ödevde ve projede Matlabda veya Excel yapmanızı tavsiye ettik. Manyetik dizaynda tek hesaplamada optimum sonuca erişemeyebilirsiniz. Bazı parametreleri değiştirmeniz gerekebilir. O yüzden kısa bir script yazarsanız tekrar tekrar uğraşmazsınız. Ama hesabını çok karıştırıyorsa önce kabloyu seçebilirsin dediği gibi. Sadece raporda belirt ne yaptığını karışıklık olmasın sonra.

burakkemalkara, 2020-04-04:
Teşekkürler hocam

closed 2020-04-05 by tokgozfurkan