odtu/EE464 #17: Error compensators

MERTAYDIN46, 2020-04-12:
Hocam merhaba
Biz kontrol derslerinde compensator olarak lead, lag ve , lag-lead'i gördük ve phase compensation yapacağımız zaman open loop transfer function üstünden bütün işlerimizi hallediyorduk mesela plant'in phase margin'i 30 derece artırmak istesek gain cross-over frekansa gelen noktada compensator ın phase'inin 30 derece olması yeterli deyip ona göre compensator tasarlıyorduk 
fakat Type 1,2,3 ün uygulama şekli biraz farklı geldi özellik le "TI" ın dökümanlarında type-2' ye baktığımız zaman net phase'i negative olduğu için voltage converter larda kullanılmaz diyor (katılıyorum) ama derste işlerken phase boost'u olduğu için phase margini artırmak için kullanabiliriz dedik şimdi benim anlamadığım şu ki Type-2 nin phase response'unda boost olduğu doğru ancak buna rağmen değeri negative. Bu phase'i plant'e eklediğimiz zaman phase margini daha da kötü yapacağız gibi duruyor? Acaba "Type 1,2,3" 'ü kullanırken closed loop üstünden mi düşüneceğiz eğer öyleyse eskiden öğrendiğimiz bilgiler neden çalış mıyor tam anlamadım hocam.
Uzun oldu biraz fakat kafam çok karıştı. 
iyi günler.

furkankarakaya, 2020-04-12:
Merhaba Mert, bu konular oldukça kafa karıştırıcı. Aslında ödevde hedefimiz de biraz kafa yordurup daha iyi anlaşılmasını sağlamak.

Normalde G transfer fonksionlu bir sistem H ile negatif feedback alarak regüle ediliyorsa, closed loop transfer fonksiyonu(CLTF) G/(1+GH) olur. Crossover frequency gain'in 1 olduğu nokta dolayısıyla bir sistem steady-state duruma geldiyse gain'i 1 olmalı ki sürekli kendisini üretebilsin, yani steady-state durumunu koruyabilsin. Aynı zamanda şunu da sağlamak gerekir, CLTF'in crossover frequency'de gaini sonsuz olmamalıdır. Bir başka deyişle CLTF'in karakteristik fonksiyonu yani 1+GH sıfıra eşit olmamlıdır. Bir başka deyişle GH yani open loop transfer fonksiyonu(OLTF) -1'e eşit olmamalıdır. Yani aslında OLTF üzerinden sistemin kontrolünü yorumlamak mümkündür. OLTF -1'e ne kadar yakın olursa CLTF'nin karakteristik fonksiyonu sıfıra o kadar yaklaşacak dolayısı ile ufak değişiklikler büyük sonuçlara yol açacak bir başka değişle step response sonucunun overdamped mi kalacağı yada underdamped mi olacağı OLTF ile ölçülebilir. Burada da zaten phase margin dediğimiz konsept işin içine giriyor ve iyi bir damping oranı yakalamak için phase margini genelde 45-60 arası bir noktada tutuyoruz.

Lag, lead veya lag-lead compensatorlar aslında H'nin fazı nasıl değiştirdiğine göre verilen isimler.  Lead type compensator (PD compensator) fazı daima pozitifte olduğu için bu ismi alıyor. Pole ve zeroyu birbirinde yeteri kadar uzak tutarsan sistemin fazını 90 derece kadar lead konuma getirmen mümkün. Benzer şekilde Lag type compensator (PI compensator) fazı daima negatifte olduğu için bu ismi alıyor ve 90 derece kadar fazı aşağıda kaydırabiliyorsun. PID compensator veya lead-lag compensator ise fazı bazı frekans aralığında lead konuma bazı bögelerde de lag konuma doğru taşıyor. Bizim yayın olarak kullandığımız type compensatorler ise bu alanda özel olarak isimlendirilmiş lag, lead, lag-lead compensatorlarden başkası değil.

Şimdi hangi type compensator seçmen gerektiği ise sisteminin stabil hale getirmek için lead edilmeye mi ihtiyacı olduğu yoksa response hızının artırılması için lag edilmeye mi ihtiyacı olduğu ile ilgili. Bu da doğrudan sistemin 'G' pole ve zero konumu dolayısı ile bode-plot'u ile alakalı. Yaklaşımın benim sistemimin G şeklinde bir transfer fonksiyonu var, bu sistemin lag edilmeye veya lead edilmeye veya bölge bölge ikisine de ihtiyacı var gibi bir sonuca varıp, compensator typelardan senin için gerekli olana karar vermek.

ODTUClass'a koyduğumuz Fundamentals of Power Electronics kitabının 9. chapterına göz atmanı tavsiye ederim, gerçekten iyi bir kitap.

MERTAYDIN46, 2020-04-12:
Hocam çok teşekkür ederim.

MERTAYDIN46, 2020-04-20:
Hocam bir sorum daha olacakta hakikaten çok kafa yordum anlamadım.
Şimdi hocam bu compansatörler opampla yapıldığı için bir -180 derece shift geliyor zaten daha sonra denominatorda bir "s" olduğu için -90 da oradan geliyor bunlar -270 dereceden başlıyor lar phase responsa yani +90 diyelim. type-2 en fazla 90 derce boost veriyor type-3 de 180. fakat zaten benim elimde +90 olduğu için her ikisindede benim en düşük verebileceğim phase bu zaten ben burada 70 derece phase'e ihtiyacım olsa bunları kullanamam ki zaten ?  
bir de gönderdiğim ekte phase margini -360 dereceden hesaplamış hala niçin anlamadım hani desem bu opamp'tan gelen-180 işin katmış ama değil çünkü onu zaten compansator a katmış 

yardımcı olursanız çok sevinirim hocam sağolun şimdiden

[image]

furkankarakaya, 2020-04-20:
Merhaba Mert, soruyu anladığım kadarıyla cevaplamaya çalışacağım. Eğer aradığın cevap bu değilse soruyu biraz daha açarak sorar mısın?

Şimdi compensatorların fazlarının başlangıcını belirleyen şey transfer fonksiyonu. Yani op-amp ile compensator kuruyoruz diye kafadan 180 derece geliyor demeyelim. Opampı içeren girişi outputtan gelen gerilim, çıkışı duty cycle olan bir compensatordan bahsediyoruz. Kompensatorun transfer fonksiyonunu, Gc(s) veya H(s), bulduğumuzda zero pole varsa faz -90 dereceden başlar. [Şu app.note'taki Type II compensator bunun bir örneği.](http://www.ti.com/lit/an/slva662/slva662.pdf) Power supply'ın kendi transfer fonksiyonu, Gp(s) veya G(s), sorunun içerisinde verdiğimizi göz önüne alırsak eğer 1 zero, 1 RHP zero, 2 pole içeriyor. Ve zeronun frekansı yüksek olduğu için faz 0'dan başlayacak bir süre düşecek ve sonra az bir miktar zeronun etkisiyle artacak (ödevdeki üçüncü app.note). Dolayısı ile G(s)'de de +90 derece bir durum olmayacak. Böyle bir durumda Type II veya Type III işini görecek diye düşünüyorum. Loop gain'i tanımlamak en önemli kısım.

Attığın örneğin tam olarak ne olduğunu bilmiyorum, eğer HG OLTF ise 360 dereceden bakılması bana mantıklı gelmedi.

MERTAYDIN46, 2020-04-20:
Hocam teşekkür ederim
Ama ben dediğiniz şeyleri anladım onlar sıkıntı değil, sıkıntı olan şu ki her döküman farklı bir şeyler yapmış. Söyle size bir önceki gönderdiğim ekin bulunduğu slaytın linki
https://www.onsemi.com/pub/Collateral/TND352-D.PDF
buranın 34.slaytında. Evet HG open loop. Şunu sormak istedim ki open loop un phase marginini -180 den ölçen var -360 'ın aşağısından ve yukarasından ölçen var. Benim anlamadığım bazen bu opamptan gelen (-)'yi compensetor lara katıyorlar buna rağmen phase margini -360 derecen ölçüyorlar yani biz -180 den ölçüyoruz normade fakat o (-) den dolayı -360'a kaydırıyorlar. Benim anlamadığım bunu oraya kaydırıken aynı zamanda o (-) den gelen phase i compensatore de ekliyorlar. işte bu yüzden hesap yapamıyorum
umarım anlaşılır olmuştur hocam. kafam çok karıştı anlatamamış olabilirim yine.
iyi günler dilerim.

closed 2020-05-07 by furkankarakaya