odtu/EE464 #18: transfer function karışıklığı

buryalcin, 2020-04-19:
Hocam beni slaytlara yönlendirdiniz ama açıkçası aradığım cevabı bulamıyorum. Slaytlarda elde edilen tf de V_o(s) / d(s) için ve Vs pure dc sayılmış. Laplace domainde zaten bir input bir output için tf yazılır, bu mantıklı gayet. V_o/V_d bulacaksak d sadece pure dc (D) gibi kabul edip mi çözmeye çalışacağız ?

Açıkçası control-to-output / input-to-output ne demek onları da tam anlamış değilim, bu terminolojiye de denk gelmedim hiçbir yerde. Kast edilenler doğru mudur ondan da emin olamıyorum.

furkankarakaya, 2020-04-19:
Merhaba Burak, sorduğun soruyu ödevde biz sorduğumuz için doğrudan cevap vermek zor. Senin araştırman gerekiyor. Terminoloji kesinlikle doğru ve internette onlarca sonuç var. [Control-to-output](https://www.google.com/search?q=control+to+output+transfer+function&oq=control+to+ou&aqs=chrome.0.69i59j69i57j69i59l2j0l4.1998j0j7&sourceid=chrome&ie=UTF-8). Odtuclass'a yüklediğimiz kitabın 8. chapterı da bu konuları işliyor, ondan da faydalanmanı tavsiye ediyorum. 

Sorunun ilk kısmı da small signal model ile ilgili. Small signal model zaten bir değişkenin ortalamasını ve bu ortalama üzerinde nasıl dalgalandığını içeriyor. Dolayısıyla elektriksel olarak hem DC hem de AC kısmı olmalı ki small signal modelin karşılığı olsun. DC kısımlar devrenin operating pointini bulmak için kullanılıyor, AC kısım da bu operating pointin etrafında nasıl bir dalgalanma olduğunu anlamak için. Belki small signal analysis üzerine birşeyler okuman faydalı olabilir. 

DC/DC converter'ı multi input single output sistem şeklinde düşünebiliriz. Giriş gerilimi ve duty cycle inputlar, çıkış gerilimi de output olur bu durumda. Dolayısı ile output'tan iki input'a da ayrı transfer fonksiyonu yazılabilir ve bu iki input lineer sistemde superposition ile ele alınabilir.

buryalcin, 2020-04-19:
tamamdır hocam ben çözdüm sanırım meseleyi, teşekkürler

closed 2020-05-07 by furkankarakaya