odtu/EE464 #20: Linearization of converter in matlab.

MERTAYDIN46, 2020-04-26:
Hocam merhaba, transfer function kullanabiliyor olmamıza rağmen ben yine elimde bir tool olsun diye simulink in linear analysis toolundan converter'ı operating pointte lineerleştirmeye çalıştım açıkçası çok yaklaşmama rağmen bir türlü ac equivalent modelde bulduğum response u ve polleri bulamıyorum. yaptığım şeyi size söyle özetleyeyim ekte gönderdiğim modelden duty cycledan perturbation veriyorum outputum zaten vurguladığım gibi open loop output olarak belirledim. sonra linearization tool una açık step response una bakıyorum aşırı derecede büyük dc gain geliyor ve poleler bir nevi yakın olsa da pek te elimle derive ettiğim transfer function daki pollerle alakası yok 2.nci ekte gönderdiğim resim matlabın linear analys tool u. ben burada linearize edeceğim noktayı seçiyorum örneğin initial conditon veya snap-shot deyip devreyi istediğim zamandaki değerlerine göre linearize edebiliyorum her türlü toolu denedim Aslında operating pointleri güzel buluyor fakat neden gerçektekiyle farklı transfer functionlar alıyorum hala anlamadım. Aslında birebir aynısı gelmesi lazım çünkü bizim ac equivalent tan derive ettiğimiz transfer function zaten linearization sonucunda gelen bir şey. Acaba bu tool u kullanırken gözümden kaçan bir şey mi var hocam 

[image]
[image]

furkankarakaya, 2020-04-28:
Mert merhaba,

Issue'yu gördüm ama şu sıralar elim yoğun olduğu için dönemiyorum. Zaten ödevde de istediğimiz birşey olmadığı için biraz bekleteceğim.

MERTAYDIN46, 2020-04-28:
Tabi hocam, düşünmeniz yeter :)

closed 2020-05-10 by MERTAYDIN46