odtu/EE464 #21: compensator design

buryalcin, 2020-04-29:
Hocam merhaba,
Compensator tasarlarken önemli olan frekanslar hep birbirine çok yakın. Type-3 compensator kullanıyorum boost converter ile çalıştığımız için ama pole-zero yerleri birbirine o kadar yakın ki o bildik type-3 plotu bile çıkmıyor. Switching frekansı resonant frekansa çok yakın, nereye crossover'ı yerleştirsem compensator düzgün iş yapmıyor. Üzerinde çalışılabilecek aralık çok düşük, phase plotu çat diye iniyor aşağı compensator gelince. Ben bu işin içinden çıkamadım.  

Verilen değerlerde ya da switching frekansında bir yanlışlık olmuş olabilir mi acaba ?

buryalcin, 2020-04-29:
switching frekansı resonanta yakın demişim, o yanlış. baskılamaya çalışırken bölü 5-10 demiştik, o değer resonanta yakın demek istemiştim.

furkankarakaya, 2020-04-30:
Merhaba Burak,

Evet dediğin doğru pole ve crossover frekans birbirine yakın geliyor, tasarımı biraz daha zorlaştırdığını söyleyebilirim ama sen sorunca ben birşeyler tasarlayıp devreyi kurdum. Sonuç alabildiğimi görüyorum. Aşağıda full load to half load transition waveformu var. Sanırım harika bir tasarım olmadı ama iş görecek sonuçlar alınabiliyor gibi görünüyor. Biraz daha uğraş istersen. Çıkamazsan işin içinden pole zero değerlerini gönder, bir bakalım.
[image]

OgunAltun, 2020-05-02:
Furkan Hocam merhaba,

Attığınız grafik simulinkten sanırım. Ben compensatoru simulinkte nasıl kuracağımızı bulamadım opamp olduğu için. İnternette aradığımda converterlar için PID kontrolcü örnekleri buldum sadece. Compensatoru simulinkte nasıl implement edeceğimiz konusunda yardımcı olabilir misiniz?

furkankarakaya, 2020-05-02:
Merhaba Ogün,

Ben şu parçayı kullanarak tasarladım. Çevresini de compensatordaki direnç kapasitörler ile doldurmak gerekli.
https://www.mathworks.com/help/physmod/simscape/ref/opamp.html

Bu parçayı kullanırsanız mavi komponentler ile siyah komponentleri birbirine bağlayamazsınız. Ya hepsini mavi komponentler ile kurabilirsiniz ya da sadece kompensator kısmını kurup sinyalleri Simulink-PS veya PS-Simulink blokları ile dönüştürüp taşıyabilirsiniz.

OgunAltun, 2020-05-02:
Teşekkürler hocam şuan LTspice çözümünü buldum gibi ordan devam edip olmazsa bunu denerim.

closed 2020-05-07 by furkankarakaya