odtu/EE464 #23: Pwm generation

MERTAYDIN46, 2020-05-09:
Hocam merhaba aşağıdaki resmi Ozan hocanın slaytlarından aldım.
biz bu control sinyallerini bir Ts süresi kadar bekletip daha sonrada genişletip daraltıyor muyuz?
animasyonu izledim ama kontrol sinyalleri continous bir şekilde değişiyor gibi görünüyor. Ancak benim anladığım kadarıyla herhangi bir anda snapshot aldığımızda switching sequence 000-001-011-111 ise bu sequence'un tamamlanıp ondan sonra control signallerini discreet bir şekilde artırmamız veya azaltmamız lazım değil mi? 

[image]

MERTAYDIN46, 2020-05-10:
Bide hocam şunu sormak istedim aslında biz vektörleri(V1 ve V3) sanki aynı anda oluşturutuyormuşuz gibi vektörel toplam yapıyoruz ama esasında Voltajı farklı zaman ve farklı coordinatlarda veriyoruz yinede load da inductor olduğu için time avaragından amplitude'un ayarlanmasını anlıyorum ama sonuçta birde iki tane farklı coordinatta flux var bunların space avarage'ını alacak bir component yok gibi sanki ondan dolayı aslında baya bir torque ripple mız olacak değil mi hocam doğru anlamış mıyım?

furkankarakaya, 2020-05-10:
Merhaba Mert,

İkinci yorumundan başlayalım, dediğin doğru. Motorun endüktansı filtre gibi davranarak, uyguladığımız vektörlerin etkilerinin anında ortaya çıkmasından öte bir toplam etki yaratmasını sağlıyor. Devamını tam anlayamadım ama sanırım farklı vektörler farklı excitationlar yaratarak bir periyotta farklı fluxlar yaratıyor bu yüzden tork yüksek gelir gibi yorumladığını anlıyorum. Motorda tork asıl olarak akım ile değişir, yukarıda bahsettiğimiz gibi akım motor endüktansında süzüldüğü için o kadar şiddetli bir ripple oluşmaz.

İlk kısım için ise herhangi bir sektördeki referans vektörü aslında belirli vektörlerin sürelerini ayarlayarak oluşturuyoruz. Anahtar pozisyonlarını carrier frekansı ile samplelıyoruz. Yani bir referans vektörü yaratmak için hangi vektörlerden hangi sürelerde uygulanması gerektiğine karar verip bir sonraki carrier periyodunda onu uyguluyoruz. O arada referans vektör kayıyor tabii ama bunu dikkate almıyoruz. Sonra güncel pozisyon ile bir sonraki vektör hesabını gerçekletirip onu uyguluyoruz.

MERTAYDIN46, 2020-05-10:
1.ncisini çok iyi anladım hocam sağolun. 2.ncisinde demek istediğim şu aslında mesela A fazını 0.9 saniye, hemen yanındaki -C fazını 0.1 saniye excite edince kaba taslak A fazının 6 derece yukarsında bir MMF vektörü oluşturmayı bekleriz değil mi? fakat aslında oluşturduğumuz bu MMF'in lokasyonu yani geometrik olarak, peak değerinin açısal olarak konumu 0.9 saniye için 0 derecede 0.1 saniye için 60'ıncı derecede olacak yani bir zaman dilimi için 0'ıncı derecede, bir zaman dilimi için 60'ıncı derecede , MMF oluşturuyoruz. bu ikisinin hangi süreyle varolduğu neden hangi açıda görüneceğini etkilesinki? inductance bu açı farkını nasıl filtreliyor anlamadım sonuçta inductance farklı time larda verilen farklı current ların ortalamasını alıp ona göre bir average akım oluşturabilir ama burada farklı zamanda verilen Akımlardan başka bide üstüne farklı lokasyonda verilme durumu var.

MERTAYDIN46, 2020-05-10:
Aslında şöyle düşündüm hocamda, mesela elimizde bir senkron makine olsun ben bu makinenin rotoruna motoring mode da 6 derece load angle bir stator field uygulamak istiyorum. rotorun tam A fazı ile align olduğu anda( tabi steady-state olarak yani rotor'un senkton speed döndüğünü varsayıyorum) yukarıda bahsettiğim örnekteki gibi 0.9 saniye A fazını, 0.1 saniye de -C fazını excite edince flux 6.derecede oluşuyor gibi oluyorsa anladım hocam.

furkankarakaya, 2020-05-10:
Evet 6. derecede oluşuyor gibi oluyor. Çünkü carrier frekansı motorun dönüş hızına göre yüksek dolayısı ile bir carrier periyodu içerisinde motoru sabit gibi düşünmek gerekli. İki farklı açıdan flux yaratmak aslında bu iki vektörün süreleriyle çarpılıp toplamı şeklinde etki etmesini sağlıyor. Space vektörün temel olayı bu zaten, yaratabildiğin vektörler ile normalde yaratamadığın vektörleri elde etmek.

MERTAYDIN46, 2020-05-10:
Sağolun hocam :)

closed 2020-05-12 by furkankarakaya