odtu/EE464 #24: Hardware Project - Transfer Function

OgunAltun, 2020-05-24:
Hocam merhaba
Hardware projesinde transfer fonksiyonu çıkarma kısmı için kitapları biraz araştırdım. Erickson forward converter için buck converter transfer fonksiyonunu turns ratio ile çarparak kullanabiliriz yazmış. Biz de derivation kısmında buck converter için derive edip turns ratio yu daha sonra fonksiyona dahil edebilir miyiz ?

furkankarakaya, 2020-05-25:
Merhaba Ogün, Erickson'un ele aldığı devre ideal bir devre diye düşünüyorum. Dolayısı ile orada direkt çarpım sonuç getiriyor olabilir ama ideal olmayan bir devrede birebir çarpamayız diye zannediyorum. Tercih sizin, non-ideal olarak derive edebilirsiniz, herşeyi ideal olarak ele alıp devam edebilirsiniz ama en son hesapladığınız noktaya ulaşamayabilirsiniz, internetten örneklerden faydalanabilirsiniz belki. Günün sonunda önemli olan hedeflediğiniz tasarımı ne kadar gerçekleştirebildiğiniz ve gerçekleştiremediğiniz noktaların sebeplerini anlamış olup olmadığınız.

OgunAltun, 2020-05-25:
Evet hocam orada her şeyi ideal düşünmüş. Ben capacitor ve inductor için ESRı dikkate alarak yeniden derive etmeyi düşündüm soruyu sorarken ancak bu durumda transformerı ideal kabul etmiş oluyorum yine. Her şeyi ele alarak yapmaya çalışıp olmazsa dediğiniz gibi bazılarını ihmal edip sonucu yorumlamayı deneyebiliriz. Cevabınız için teşekkürler.

closed 2020-06-04 by furkankarakaya